emfcamp/badge-2024-software #13: Enable Bluetooth support

robmckinnon, 2024-05-02:
Can we please consider enabling Bluetooth support in the MicroPython firmware for the badge by default?

What are the arguments for or against this? 

Is it a power consumption issue, is there a way around that?

I'd like to write an app for the badge to use [BLE MIDI](https://midi.org/category/midi-connections/bluetooth-low-energy-ble-wireless-midi) to communicate wirelessly with BLE MIDI enabled synth and MIDI controller hardware. I imagine this would be easier to develop and share with others if Bluetooth was enabled by default.

In the [mpconfigboard.h](https://github.com/search?q=repo%3Aemfcamp%2Fbadge-2024-software%20mpconfigboard.h&type=code) file adding `#define MICROPY_PY_BLUETOOTH                (1)` enables Bluetooth support in the MicroPython firmware.

MatthewWilkes, 2024-05-20:
@robmckinnon BLE support is already enabled, but the full Bluetooth stack isn't. Is that sufficient for you?

At https://github.com/emfcamp/badge-2024-software/blob/main/tildagon/mpconfigboard.cmake#L39 we include the following config file: https://github.com/micropython/micropython/blob/cfd5a8ea3a80930a79dbae6a6aff54587393d0b8/ports/esp32/boards/sdkconfig.ble

I can confirm that `bluetooth.BLE` is available on our test builds, and we've been able to view advertisements.  Does that cover your needs?

fridgehead, 2024-06-02:
yeah it works for us too, using aioble to talk to a strip of LED's in our dome:
```

async def connectDevice(self):
        device = aioble.Device(aioble.ADDR_PUBLIC, "23:00:00:00:00:00")
        conn = await device.connect()
        try:
            connection = await device.connect(timeout_ms=2000)
        except asyncio.TimeoutError:
            print('Timeout')
        print("connected!")
        temp_service = await connection.service(bluetooth.UUID(0xafd0))
        temp_char = await temp_service.characteristic(bluetooth.UUID(0xafd1))
        print("offen")
        offbytes = b"\x5B\x0f\x00\xB5"
        await temp_char.write(offbytes, timeout_ms=1000)

```