emfcamp/badge-2024-software #16: Widget library

hughrawlinson, 2024-05-04:
### Widgets we could have
- [x] Dialog
- [x] Notification
- [ ] Menu
- [ ] NestedMenu
- [ ] Dial?
- [ ] Label/Text?

Should we have some sort of theming option too so the badge UI and apps that choose to use these widgets look "on brand"?

andypiper, 2024-06-11:
I think a text block / wrapped text widget would be very useful. My efforts to get a block of text to wrap nicely based on `ctx` APIs have been challenging (or, my understanding of it is incomplete).

hughrawlinson, 2024-06-12:
I believe we actually have a text block now - @MatthewWilkes worked on it - but it's not documented yet. I'll open an issue.

npentrel, 2024-06-16:
Text wiget is now documented in [layouts](https://tildagon.badge.emfcamp.org/tildagon-apps/reference/ui-elements/#layouts)