emfcamp/badge-2024-software #21: Startup Logo and Sponsor Logos

hughrawlinson, 2024-05-06:
We've got sponsor logos to show on badge startup. This applies to all gold and badge sponsors in the sponsorship sheet. @jna_sh can you let us know if there are any specific requirements around timing, prominence, etc?

joenash, 2024-05-06:
No specific requirements around timing. Prominence wise, we don't currently have a sponsor above Gold, so they can all be equally placed. It might be nice to do two screens of logos, one for the Gold sponsors, and then one for the badge sponsors that says "badge sponsors" or something at the top. See https://emfcamp.org/sponsors for example.

andypiper, 2024-06-11:
Presumably this could now be closed, as no longer valid (for 2024 at least)?

npentrel, 2024-06-16:
I think it would probably be nice to still do this, now that images work, so let's keep it open