emfcamp/badge-2024-software #32: Hardware APIs

hughrawlinson, 2024-05-18:
I'm not sure which hardware is on the device beyond the accelerometer, but we should decide how we want app writers to interface with it - I know it's fairly well documented for hexpansions, but onboard hardware would be good.

If I understand correctly we have a couple of options:

1) The hardware is already connected over the i2c mux. We document how to use the i2c mux and which devices are at which addresses, and leave app authors to it
2) We write little driver apps to talk to the hardware over i2c and translate messages into an event bus based interface. We then document these interfaces, and in the hexpansion guide, for hexpansion devs who are attaching i2c hardware, we tell them to provide similar event bus based interfaces to the hardware
3) We have micropython classes/objects that we provide in a library to app authors, that under the hood speak either i2c or over the event bus to the hardware.  We encourage hexpansion authors to provide these classes for app authors to copy paste into their apps?

If someone could add a list of all the hardware, that would be cool

ChrisDick, 2024-05-18:
The power module has the following API covering the PMIC and USB C ports:
import power
BatteryChargeState()
    battery charge status - "Not Charging", "Pre-Charging", "Fast Charging", "Terminated"
BatteryLevel()
    charge state estimate (%.)
Fault()
    Battery, charge and boost fault states.
Icharge()
    Battery charge current (A.)
Off()
    Turn the badge off, if plugged into USB, badge will turn off when disconnected.
SupplyCapabilities
    reports the capabilities of the power supply attached.
Vbat()
    Battery voltage (V.)
Vin()
    Input voltage (V.)
Vsys()
    system voltage (V.)
Enable5V(enable)
    turn the 5V usb out supply on or off.

The IMU we haven't looked at as we changed it last minute and no one has one to test. likely follow the fl0w3r.

The port expanders will be part of the hexpansion and top board APIs 

There is no direct access to the i2c mux, the fusb of the top board channel of the mux or the system channel.

ChrisDick, 2024-05-18:
for reference the Flow3r IMU API is
acc_read()
x, y, z readings
gyro_read
x, y, z readings
pressure_read()
pressure and temperature readings

Molive-0, 2024-05-26:
If it is possible, having the pedometer interface would be something I am interested in.

npentrel, 2024-06-16:
I think we can close this now, right @hughrawlinson ? For the pedometer, there is already a PR up

hughrawlinson, 2024-06-16:
I think there's still cool stuff to be done here - if we've got interfaces to speak to all the built in hardware (I'm not sure, I don't know the hardware stuff very well), it would be cool to see if we can find a way to let hexpansion makers expose their hardware (sensors, whatever else) in a similar way to app programmers. Then, maybe we want a hardware availability API. But this issue isn't well specified, feel free to close it!

npentrel, 2024-06-16:
Happy to leave open - I thought this was about onboard hardware which I think we now fully cover - but sounds like this goes beyond that