emfcamp/badge-2024-software #34: Text Widget

hughrawlinson, 2024-05-18:
We could provide a text widget that takes a string and automatically breaks it into lines so that it fits in the centre of the screen given a certain line height. If we wanted to get even fancier, we could let the user specify a target centre point, and have the widget fit the text within the screen boundaries as close as possible.

MatthewWilkes, 2024-05-29:
Done in v0.0.7

MatthewWilkes, 2024-05-29:
```
import app_components.layout
text = app_components.layout.TextDisplay("Lorem ipsum " + "abcde "*30)

text.draw(ctx)
```

Alternatively, to allow scrolling:

```
import app_components.layout
text = app_components.layout.TextDisplay("Lorem ipsum " + "abcde "*30)
layout = app_components.layout.LinearLayout([text, foo, bar])

layout.draw(ctx)
```
and manipulate either `layout.y_offset` or pass button up and down events to `layout.button_event`

closed 2024-05-29 by MatthewWilkes

andypiper, 2024-06-13:
Am I correct in thinking (finding) that this does line wrapping, but does not do word wrapping?