emfcamp/badge-2024-software #37: Calling wifi.get_ip throws

hughrawlinson, 2024-05-18:
Seems like the API doesn't quite line up somewhere. This is after confirming that `wifi.isconnected()` returns true (which is weird because I'm very unsure how the badge would connect to our wifi, I would expect it to return false)

```
  File "/home/hugh/Development/github.com/emfcamp/badge-2024-software/modules/wifi.py", line 191, in ifconfig
    return _STA_IF.ifconfig()
AttributeError: 'WLAN' object has no attribute 'ifconfig'. Did you mean: 'config'?
```

sknebel, 2024-05-30:
works right now (with current firmware, in the actual field). If not connected, `None` is returned.

andypiper, 2024-06-11:
I think this can now be closed?

npentrel, 2024-06-16:
> I think this can now be closed? (happy to support issue triage/labelling if the team would like help on that)

I just started doing that @andypiper - please feel free to as well though!

npentrel, 2024-06-22:
This no longer throws, if the issue recurs we can open a new issue.

closed 2024-06-22 by npentrel