emfcamp/badge-2024-software #87: ctx.image silently fails

nicl83, 2024-06-01:
The `ctx.image` function appears to be broken in some way on the latest firmware (V1.3.0 at time of writing). I've specified an absolute path, and tried both GIF and PNG format images, but nothing is actually drawn to the screen.

```py
def draw(self, ctx):
    ctx.image("/apps/myapp/qr.png", -58, -58, 116, 116)
```

No exception is thrown on the `mpremote` console, but nothing is drawn to the screen. I'll poke at the source later and see if it's an obvious issue, but I just wanted to log this somewhere before I forget...

bcc, 2024-06-01:
I asked about this on IRC earlier, it's not working but coming soon: "we're hoping to fix this with an OTA sometime in the near future but probably not today".

npentrel, 2024-06-22:
This is now fixed in the latest version

closed 2024-06-22 by npentrel