emfcamp/badge-2024-software #90: Eeprom format instructions don't work on M24C16

BartDeWaal, 2024-06-01:
I'm trying to format the Eeprom according to instructions here:

https://tildagon.badge.emfcamp.org/hexpansions/eeprom/

I'm using the 1.3.0 version, with the following adjustments:
```
header = HexpansionHeader(
    manifest_version="2024",
    fs_offset=32,
    eeprom_page_size=16,
    eeprom_total_size=1024 * (16 // 8) // 8,
    vid=0x0023,
    pid=0x0001,
    unique_id=0x0,
    friendly_name="MCP3021",
)
```

It gives the following output:
```
~/p/t/badge-2024-software> mpremote mount modules + run modules/scripts/prepare_eeprom.py
Local directory modules is mounted at /remote
Detected eeprom at 0x50
Using 1 bytes for eeprom internal address
Writing 16 bytes at b'\x00': b'THEX2024 \x00\x10\x00\x00\x01\x00\x00'
Writing 16 bytes at b'\x10': b'#\x00\x01\x00\x00\x00MCP3021\x00\x00I'
Warning: possible unsupported chip. Size: 256
8 chips detected. Total EEPROM size 2048bytes.
Page size: 16
eeprom block count: 4
partition block count: 0
partition block size: 512
Traceback (most recent call last):
  File "<stdin>", line 49, in <module>
OSError: 28
```

I've attached a logic analyzer trace.
[m24c16_write_eeprom.sr.tar.gz](https://github.com/user-attachments/files/15522126/m24c16_write_eeprom.sr.tar.gz)

This is measured from port 2, looking at the mux data (going to 0x77) it's not clear to me why I can see it on port 2.

BartDeWaal, 2024-06-01:
Actually, on closer inspection, I think it writes correctly but just gives an error

hairymnstr, 2024-06-01:
The EEPROM header has written fine but the littlefs filesystem generation failed. If you don't need to store code on the hexpansion this is fine.

BartDeWaal, 2024-06-01:
I do in fact want to write to the filesystem. Further investigation shows it's an issue with writing the littleFS.

hairymnstr, 2024-06-20:
Fixed in 1.8.0 now that littlefs formatting is working on the M24C16.

closed 2024-06-20 by hairymnstr