emfcamp/badge-2024-software #91: Issue with the menu.py

Johnr24, 2024-06-01:
I'm getting this

```
Traceback (most recent call last):
  File "main.py", line 39, in <module>
  File "system/scheduler/__init__.py", line 248, in run_forever
  File "asyncio/core.py", line 1, in run_until_complete
  File "asyncio/core.py", line 1, in run_until_complete
  File "asyncio/core.py", line 1, in run_until_complete
  File "system/scheduler/__init__.py", line 244, in _main
  File "asyncio/funcs.py", line 1, in gather
  File "asyncio/core.py", line 1, in run_until_complete
  File "system/eventbus.py", line 79, in run
  File "app_components/menu.py", line 70, in _handle_buttondown
TypeError: function takes 2 positional arguments but 3 were given
MicroPython 5114f2c-dirty on 2024-06-01; Tildagon with ESP32S3
```

when I run this code on the badge and observe with mpremote, 

if this is a problem with my code and you can fix, find me in twask village and I can give you a drink! 

```
from app import App
import asyncio
from app_components import Menu, Notification, clear_background
from events.input import Buttons, BUTTON_TYPES
from tildagonos import tildagonos, led_colours
import urequests

main_menu_items = [" ", "A", "B", "C", "D", "E", "F", "G", "H", "I", "J", "K", "L", "M", "N", "O", "P", "Q", "R", "S", "T", "U", "V", "W", "Y", "Z", "0", "1", "2", "3", "4", "5", "6", "7", "8", "9"]
class SnakeApp(App):
    def __init__(self):
        self.menu = Menu(
            self,
            main_menu_items,
            select_handler=self.select_handler,
            back_handler=self.back_handler,
        )
        self.notification = None
        self.selected_letters = []
        self.button_states = Buttons(self)
        self.lowercase_toggle = False
        self.input_sequence = ''

    def select_handler(self, item):
        if self.lowercase_toggle == True:
            item = item.lower()
        else:
            item = item.upper()
        self.selected_letters.append(item)
        sequence = ''.join(self.selected_letters)
        self.input_sequence = sequence  # update the input sequence with each letter pressed
        if len(self.selected_letters) == 4:
            self.notification = Notification(sequence + '!')
            print(sequence)
            self.send_word(sequence)
            self.selected_letters = []  # reset the list of selected letters
        else:
            self.notification = Notification(item + '!')

    def send_word(self, word):
        print(f'Word: {word}')
        wxrd = word.replace(' ', 'x')
        wxrd = wxrd.replace('-', '£')
        url = 'no uploaded'
        headers = {no uploaded}
        data = no uploaded
        response = urequests.post(url, headers=headers, json=data)
        print(f'Data sent: {data}')
        print(response)  # print the response object
        print(response.text)  # print the response from the server

    def back_handler(self):
        self.minimise()

    def draw(self, ctx):
        clear_background(ctx)
        if self.lowercase_toggle:
            self.menu.items = [item.lower() for item in main_menu_items]
        else:
            self.menu.items = [item.upper() for item in main_menu_items]
        self.menu.draw(ctx)
        if self.notification:
            self.notification.draw(ctx)
        guifont = 0
        inputfont = 0
        ctx.font = ctx.get_font_name(inputfont)
        ctx.rgb(255, 255, 255).move_to(60,0).text(self.input_sequence)
        ctx.font = ctx.get_font_name(guifont)
        ctx.rgb(255, 255, 255).move_to(-70,60).text('←DEL')
        ctx.font = ctx.get_font_name(guifont)
        ctx.rgb(255, 255, 255).move_to(60,60).text('SEND→')
        ctx.font = ctx.get_font_name(guifont)
        ctx.rgb(255, 255, 255).move_to(60,-80).text('CAPS→')
        ctx.font = ctx.get_font_name(guifont)
        ctx.rgb(255, 255, 255).move_to(60,-60).text('LOCK')
        ctx.font = ctx.get_font_name(guifont)
        ctx.rgb(255, 255, 255).move_to(-60,-60).text('QUIT')

    def update(self, delta):
        self.menu.update(delta)
        if self.notification:
            self.notification.update(delta)
        if self.button_states.get(BUTTON_TYPES["RIGHT"]):
            self.lowercase_toggle = not self.lowercase_toggle  # toggle the value
            if self.lowercase_toggle:
                self.notification = Notification('Lowercase')
                  # change menu items to lowercase
            else:
                self.notification = Notification('Uppercase')
                 # change menu items to uppercase
            self.button_states.clear()

        
        #backspace implementation
        if self.button_states.get(BUTTON_TYPES["LEFT"]):
            self.selected_letters = self.selected_letters[:-1]
            self.input_sequence = ''.join(self.selected_letters)
            self.notification = Notification('Backspace')
            self.button_states.clear()

__app_export__ = SnakeApp
```

hairymnstr, 2024-06-01:
There's a docs bug here. The `select_handler` function needs a second parameter item_index which you probably won't use. 

    select_handler(self, item, item_index):

Johnr24, 2024-06-01:
Thank you. Please find the person in the orange leather jacket tonight in
or near null sector to claim your prize.

I'm also around til Monday!