emfcamp/badge-2024-software #100: Prevent main menu/launcher from constantly being drawn

enteryournamehere, 2024-06-03:
In my app, a variant on Snake, I'd like to draw a background grid. Clearing the screen and drawing this grid inside every `draw()` call takes a while, and results in `update()` only running every 170 or so milliseconds, which is too slow to handle user input.

The grid doesn't change throughout the game, so I only really need to draw it on app startup, after that the updates to the canvas are small (just the head/tail of the snake).
However, if I don't call `clear_background()` and re-draw the grid every time `draw()` is called, it is replaced by the main menu (Launcher) UI.

Is it possible for my app to stop the Launcher from being drawn, perhaps by minimising it? Perhaps this could even happen automatically for apps with a foreground service. Although it would of course also require the Launcher being brought back to the foreground once the other app is closed.

MatthewWilkes, 2024-06-03:
Agreed, I came across this during camp, also. We should change the draw loop to only do foregounded and always on top, I think.

npentrel, 2024-06-22:
@MatthewWilkes can this be closed? You've fixed this right?

MatthewWilkes, 2024-06-23:
Yes, thanks!

closed 2024-06-23 by MatthewWilkes