emfcamp/badge-2024-software #106: Add a way to get TildagonOS build information

andypiper, 2024-06-05:
Currently:

```
>>> import sys
>>> mpyinfo = sys.implementation
>>> mpyver = f'{mpyinfo.name} {mpyinfo.version[0]}.{mpyinfo.version[1]}.{mpyinfo.version[2]} {mpyinfo._machine}'
>>> print(mpyver)
micropython 1.23.0 Tildagon with ESP32S3
```

I don't see a method to get more detailed build information out of the `tildagon` class, or any other build information - this might be useful. Generally I am thinking a settings -> About screen that could display this sort of data would be a useful addition.

ChrisDick, 2024-06-24:
there's the ota API that exposes the version from esp_ota_get_running_partition().
>>>import ota
>>> ota.get_version()
'v1.8.0'

will raise a doc issue to add this

ChrisDick, 2024-08-15:
This is now in the docs

closed 2024-08-15 by ChrisDick