emfcamp/badge-2024-software #107: Text dialog with on-screen keyboard

janion, 2024-06-05:
It took me about 10 minutes to enter my home WiFi credentials. An on-screen keyboard would be much easier to use. 

I've started writing one to replace the existing text dialog. Is that a useful contribution, or would it be duplicating work? I couldn't see any other issues, branches or PRs mentioning a keyboard.

emericklaw, 2024-06-05:
You might want to checkout the keyboard on the flow3r. 

Might be slightly more challenging on the EMF badge with only 6 (7if you count boop) buttons.

This was brought up a few days ago in the badge channel.

https://matrix.to/#/!VvSoIxosxjlkTNMLOa%3Amatrix.org/%24TF0Zu9ifF70dmdz6y-RpURiTQMxrmXeWCk6ty30X1QM

[image]

janion, 2024-06-06:
@emericklaw That link to the badge channel didn't work for me. It said that the link was malformed.

I see that a new keyboard PR has just been opened, but I have a demo app of what my keyboard would look like: https://github.com/janion/keyboard-demo.

Obviously I'm biased, but I think that my keyboard is more intuitive because it's closer to a standard keyboard layout. I'll leave it up to whoever is in charge to decide which one (or both) should be published.

janion, 2024-06-06:
I haven't managed to get the simulator working yet, but here's a photo of the keyboard: 
[image]

npentrel, 2024-06-16:
We now have an on-screen keyboard as of 1.7.0, I'll change the title of this ticket to add this as another keyboard option. People could switch between the two