emfcamp/badge-2024-software #123: Request: support multiple wifi profiles

andypiper, 2024-06-09:
Unsure whether this should be an app, or built-in?

Notionally, this is a straightforward (!) as having an array of network settings in the settings, and swapping them in and out as needed. That's the basic idea I had for making an app to do this, anyway. Should it be part of the core though...

andypiper, 2024-06-18:
In the meantime I wrote [a quick script](https://gist.github.com/andypiper/127346173c67bb08219cdae3a3bfc995) to poke relevant values into the badge from outside.

RichardoC, 2024-09-04:
I'd a very quick look at the code, it looks like an app could definitely do this via wifi.connect("custom_ssid", "custom_password", "custom_username") `<https://github.com/emfcamp/badge-2024-software/blob/aa8f510a6a76e93b0f4aacce9b79663ec28e3bfb/modules/wifi.py#L90`>

RichardoC, 2024-09-05:
It's very rough and ready but I've made <https://github.com/RichardoC/wifi-switcher/tree/main> which should show up in the app store soon. Mind giving it a go and letting me know if it helps?
You can use <https://docs.micropython.org/en/latest/reference/mpremote.html#mpremote-command-edit> to edit the example file

RichardoC, 2024-09-12:
I've just updated this app to use the built in settings which should make it a bit easier to use

andypiper, 2024-09-13:
Looking forward to trying this, I don't have my badge handy but will install when I can. Thanks!