emfcamp/badge-2024-software #162: Stuck on rebooping after upgrade

edent, 2024-06-20:
**Describe the bug**
Trying to upgrade from v1.6.0 to v1.8.0 OTA.

Update downloads, then says "Status Rebooping"

LEDs stop flashing. Screen stays on.  Waited 5 minutes then manually rebooped. Device was still on 1.6.

Version: v1.8.0

**To Reproduce**
Steps to reproduce the behaviour:

1. Go to 'Update'
2. Let update commence
3. Wait 10 minutes.
4. Once device appears unresponsive, hold reboop
5. Observe update hasn't applied

**Expected behavior**
Reboot should apply updates.

Powerlines2000, 2024-06-28:
Exactly the same issue - Download seems to be fine just gets to "Reebooping" and hangs.

Jonty, 2025-02-28:
Is this still an issue? There's a new firmware release that you should flash and try.