emfcamp/badge-2024-software #175: Minimum battery voltage should be higher

TBExtent, 2024-06-24:
The minimum battery voltage is set to 2.5V, but the rated discharge cut off voltage is 3.0V for the batteries used in the badge. (see the datasheet: https://cdn.shopify.com/s/files/1/0176/3274/files/LP803860_2000mAh_3.7V_20190510.pdf?v=1665420199)

Shutting the device down when it reaches 3.0V (or even before then) would increase the lifetime of the battery.
It would somewhat decrease the usable charge, but not massively as the voltage doesn't decrease linearly with charge remaining - the voltage drops faster at low charges. 

The datasheet does list 2.5V as the overdischarge cut off voltage, but that is to prevent the battery becoming completely unusable when it is overdischarged rather than enforcing a safe minimum charge.

The batteries are also rated to be charged up to 4.2V rather than the current 4.1V max battery voltage, but that isn't really a problem unless you want to squeeze that little bit extra charge in :)

kliment, 2024-06-24:
While I agree we should change it to a higher voltage, I don't think you are reading the datasheet correctly - the battery protection IC in the battery itself is rated to cut off at 3.0V. Whether the cutoff comes from the battery or the badge is not especially consequential, but having it happen on the badge lets us handle things more gracefully in software. The 4.1V is a precaution as we support multiple different battery types, including the 2018 and 2016 badge batteries. Can you make a pull request with the proposed change?

kliment, 2024-06-24:
Never mind the above - I saw that the protection IC is in fact set to 2.5V. You're entirely correct.

ChrisDick, 2024-06-24:
The PMIC is currently set to 3.5V as a minimum voltage. We may need to increase it still as at max current (4A) the voltage drop in the TLV62595(Rdson) and inductor(RDC) is 0.312V ( 78mOhm * 4A), but that's everything pulling max current, so unlikely but possible. When the voltage drops to this level, the TLV will put the battery voltage on the output, which we don't want as it maybe too high for a 3v3 rail.
The define you want to change only effects the battery level, which I've been doing some playing around with as part of issue #83. The level for the charge termination is based on the charge current, I am going to reduce this, but the loss of stored energy is negligible in reality. During testing I've found that the battery does charge to 4.2 with the lower termination current, then drops to around 4.14V, so for max voltage when not charging it's about right. I'll likely set the min voltage around the 3.5V mark. I'll make a branch, so you can see what I'm doing.

kliment, 2024-06-24:
@ChrisDick Where is that setting in the code?

ChrisDick, 2024-06-24:
it's not, it's the default of the BQ25895. I've added the code already to the Improve-BatteryLevel branch to allow this to be fine tuned. It's on p35 of the datasheet, reg 03 - SYS_MIN.

ChrisDick, 2024-06-24:
I think the issue here is that the Vsys_min doesn't disconnect the battery, that's done at Vbat_dplz which is 2.5V. we need to add a task to monitor the battery level and turn off the badge.

ChrisDick, 2024-08-04:
fixed in v1.9.0

closed 2024-08-04 by ChrisDick