emfcamp/badge-2024-software #202: Cannot update firmware using OTA

emericklaw, 2025-05-17:
**Describe the bug**
When updating the firmware the error `ESP_ERR_HTTP_CONNECT` is shown.

Digging into this I see this:

```
<function head at 0x3c2b16e0> ('https://github.com/emfcamp/badge-2024-software/releases/download/latest/micropython.bin',) {'allow_redirects': False}
Periodic
...
Periodic
<Response object at 3c248920>
<Response object at 3c248920>
Starting thread
Periodic
In thread
<function> (<bound_method>, 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/751434638/a0f4124c-f2df-4383-9b73-bb693e9fd1e6?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=relea{}
Periodic
Exception: (-28674, 'ESP_ERR_HTTP_CONNECT')
(-28674, 'ESP_ERR_HTTP_CONNECT')
(-28674, 'ESP_ERR_HTTP_CONNECT')
```

I thought this might be related to certificate issues, however the certificate for `githubusercontent.com` has USERTrust RSA Certification Authority as its root which, if I am looking in the right place is in the list https://github.com/espressif/esp-idf/blob/v5.2.1/components/mbedtls/esp_crt_bundle/cacrt_all.pem

I altered the url here to point to a copy of the .bin on a personal website and it flashed fine so there is something weird going on with the GitHub URL.

https://github.com/emfcamp/badge-2024-software/blob/f5a3c5824acf18a229ad051996029b107b7da95f/modules/system/ota/ota.py#L153

Version: 1.9.0

**To Reproduce**
Steps to reproduce the behavior:

1. Go to 'Update'
2. Wait for update process to start
3. Error `Failed: (-28674, 'ESP_ERR_HTTP_CONNECT')` is shown

**Expected behavior**
The firmware OTA completed successfully.

dpslwk, 2025-05-17:
I believe `CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN` is set, so only certs from https://github.com/espressif/esp-idf/blob/v5.2.1/components/mbedtls/esp_crt_bundle/cmn_crt_authorities.csv are include from the full `cacrt_all.pem` you linked to

not sure if that still include the needed cert for the url to be built into the firmware

dpslwk, 2025-05-17:
https://github.com/espressif/esp-idf/issues/3218
issue mentions at the bottom the url at the end of the redirect might be to long so some buffers might need increasing?

Jonty, 2025-05-17:
If true the easiest fix is to follow redirects in ota.py - @emericklaw does it work if you change `allow_redirects` to True?

https://github.com/emfcamp/badge-2024-software/blob/f5a3c5824acf18a229ad051996029b107b7da95f/modules/system/ota/ota.py#L150

emericklaw, 2025-05-18:
I am pretty sure I have narrowed it down to certificate issues.

If I replace the https:// with http:// in the URL returned by the replease/micropython.bin redirect then the OTA update works.

https://github.com/emfcamp/badge-2024-software/blob/f5a3c5824acf18a229ad051996029b107b7da95f/modules/system/ota/ota.py#L153

```python
url = url.replace("https://", "http://")
```

---
> If true the easiest fix is to follow redirects in ota.py - [@emericklaw](https://github.com/emericklaw) does it work if you change `allow_redirects` to True?

I did try this before and it made no difference, even passing the release/micropython.bin url straight to the OTA updater still resulted in errors.

---

> I believe `CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN` is set, so only certs from https://github.com/espressif/esp-idf/blob/v5.2.1/components/mbedtls/esp_crt_bundle/cmn_crt_authorities.csv are [...]

This does include the correct root certificate but for some currently unknown reason it still errors.

---
It should be possible to get more verbose output from the `esp_http_client` and `esp_https_ota` using
```c
    esp_log_level_set("esp_http_client", ESP_LOG_VERBOSE);
    esp_log_level_set("esp_https_ota", ESP_LOG_VERBOSE);
```
But I am having a hard time getting the ESP logging to pass through to micropython.

Tomorrow I am going to setup a standalone project with just ota updates and see if the issue happens with simple code. What is also holding me back is the 5+ minute compile time, reflash, setup wifi and then I finally get to test on the badge!

Jonty, 2025-05-18:
The weird thing is that the calls to GitHub to get the path to the firmware are over HTTPS and work fine, and presumably that implementation is using the same certs?

emericklaw, 2025-05-18:
> The weird thing is that the calls to GitHub to get the path to the firmware are over HTTPS and work fine, and presumably that implementation is using the same certs?

The initial request within the Micropython OTA code is to github.com, once it redirects the new URL is 

They use different certificates but more importantly different chains (RSA vs ECC).

https://www.ssllabs.com/ssltest/analyze.html?d=objects.githubusercontent.com&s=185.199.111.133&latest
https://www.ssllabs.com/ssltest/analyze.html?d=github.com

I have added a pull request #205 to change from using the common roots to using all roots, this does increase firmware size by 46k but increases coverage from 90% to ~100%.

Looking at the certificate chain, for whatever reason the ESP IDF codes seems to want to verify the second trust path which uses **Comodo AAA Services root**, this is not included in the common certificate bundle.
If it used path 1 then the **USERTrust RSA Certification Authority** root is included.

[image]

I think this is always going to be an issue with hard coded certificates and badges that might only get updated once every two years. The Comodo root certificate expired end of 2028 so at EMF 2030 any badges with firmware from the previous year will need to be web flashed to get the renewed certificate.

MatthewWilkes, 2025-05-19:
Ah, man, I had hoped we'd sorted the jank here.

So, the reason it doesn't follow redirects is because of the huge header size, this is an issue we had on the previous badge too, where we'd end up having to allocate insanely large buffers to handle the redirect. We had trouble with the CA last time too, including the change to ECC, so had hoped that including the bundle would make the difference.

It's also the reason that the OTA call now takes a URL, in previous incarnations we did the redirect handling in C code, which made it harder to deal with some of the potential weirdness.

I don't object to us widening the CA set, and (as you say) updating the IDF will help too. That said, if replacing https with http works, maybe we should add that to the ota.py also? Well, maybe add it with a check to handle GitHub inevitably blocking http.

emericklaw, 2025-05-19:
> That said, if replacing https with http works, maybe we should add that to the ota.py also? Well, maybe add it with a check to handle GitHub inevitably blocking http.

Added HTTPS/HTTP switching on error in #206

Jonty, 2025-06-14:
Dropping a couple of other possible fixes here:
* https://forum.arduino.cc/t/https-ota-update-follow-redirects-esp32/1291198
* https://github.com/espressif/arduino-esp32/issues/9530#issuecomment-2090354891

thinkl33t, 2025-07-28:
Reopening as i'd still like to fix TLS!

PR #206 (2025-07-28): Tries OTA via HTTPS then flips to HTTP on failure then repeats
# Description

This PR changes the OTA download to first try HTTPS then on failure tries HTTP and repeats flipping between HTTPS/HTTP.

Fixes #202