emfcamp/badge-2024-software #230: Built-in namebadge app

thinkl33t, 2025-08-29:
**Is your feature request related to a problem? Please describe.**
We should have a built-in namebadge app which automatically activates when the menu has been idle for an amount of time (15s?)

**Describe the solution you'd like**

- Namebadge app
- Use 'backgrounds' so that background image matches everywhere else
- Name pulled from settings
- Maybe pronouns pulled from settings too?
- Scale text to fit the screen width

Jonty, 2025-09-02:
That's a good idea. And the settings allow you to select the active one?

MatthewWilkes, 2025-09-03:
I was wondering about doing this for a battery app, so it can launch on boot also. We could also just  have apps set themselves as the default on launch, so it feels persistent.